paulvha/scd30 #3: I2C -ESP8266 crash

prisikarm, 2019-09-09:
Hello,
I am trying to use your library on ESP8266, in a bundle with BMP280 ( pressure and temperature) and MCP9808 (temperature) sensors. After uploading the code, the ESP is stuck and starts constantly restarting.

Is it because of the clock stretching, and other sensors cannot work in this mode? If yes, how I can use all three sensors?

Thank you

paulvha, 2019-09-09:
I am on a short vacation and can look end of week. One aspect that could be is wjen bmp280 or an other library changes the speed of the i2c from 100k to higher speed ( the bmp280 can handle that)

Outlook voor Android downloaden<https://aka.ms/ghei36>

paulvha, 2019-09-11:
HI

Can you share a link to the drivers you use for the BMP280 and MCP9808 (there are multiple options available and want to look that the ones you use)

Regards,
Paul

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

paulvha, 2020-11-21:
closing assume it is solved by now

closed 2020-11-21 by paulvha