paulvha/scd30 #4: Second I2C Bus

leic3d, 2020-07-12:
Hi,

I'm trying to get a set of 2 SCD30 working with an ESP32. 
Since I did not find a way to change the I2C adress of the SCD30, I now try to use the second I2C bus of the ESP32 to get both working with the same adress. However, it seems that your twowire class in softwire does not include the functionality to use a second bus.
 I tried to compare your softwire.cpp and twi.cpp to the corresponding files of the "normal" wire.cpp (and others used) but i cannot find a solution to add the functionality to use the second bus.
Can you help with that?

Thank you very much,
Michael

paulvha, 2020-07-13:
HI Micheal,

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

paulvha, 2020-07-13:
HI

I am going to have a look at it today. I will let you know.

Regards,
Paul

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

paulvha, 2020-07-13:
I had a look and checked the latest datasheet SCD30 (march 2020). The only added option is the ability to read the firmware level. I consider adding that later. Softwire is a port of the I2C from the ESP8266 and in it's current form it only supports one I2C channel. You could try to use the standard Wire.h and set for very low speed to read the SCD30, e.g. 35Khz instead of 100Khz. You might be lucky with the clock-stretching not happening too long. For this you need to change in paulvha_scd30.h around line 64.

closed 2020-07-13 by paulvha