paulvha/scd30 #6: SCD-30 Sensor + OLED

PrateekGoyal18, 2020-10-03:
Hi, I am trying to integrate an oled with scd30 sensor over esp32, but I am getting some TwoWire errors.
My code is:
`//#include <Wire.h>
#include <WiFi.h>
#include <esp_wpa2.h>
#include <FirebaseESP32.h>
#include <NTPClient.h>
#include <WiFiUdp.h>
#include "paulvha_SCD30.h"
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define EAP_ANONYMOUS_IDENTITY "anonymous@iitd.ac.in"
#define EAP_IDENTITY "himangr.visitor@iitd.ac.in"
#define EAP_PASSWORD "epreslab@iitd"
const char* ssid = "IITD_WIFI";
int counter = 0;
WiFiClient client;
int rssi;
String notif_wifi, notif_sensor;

#define FIREBASE_HOST "labindividuallight.firebaseio.com"
#define FIREBASE_AUTH "tQNM1JXaHbrkpyVdavuwayNLnnzi0d4srHk6mptT"
FirebaseData firebaseData;

const long utcOffsetInSeconds = 19800;
String formattedDate, dateStamp, dateStamp_display, dateStamp_firebase, timeStamp, timeStamp_display, timeStamp_firebase;
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org", utcOffsetInSeconds);

#define scd_debug 0
#define SCD30_I2C_ADDRESS 0x61
SCD30 scd30;
float temperature, humidity; int co2;
String temp_str, hum_str, co2_str;

Adafruit_SSD1306 display(-1);
#define OLED_I2C_ADDRESS 0x3C
#if (SSD1306_LCDHEIGHT != 64)
#error("Height incorrect, please fix Adafruit_SSD1306.h!");
#endif

const unsigned char EPRESlogo [] PROGMEM = {
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xdc,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xdc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xdc, 0x02, 0x00,
  0x00, 0x00, 0x08, 0x01, 0xdc, 0x42, 0x00, 0x00, 0x00, 0x00, 0x41, 0x9c, 0x40, 0x00, 0x00, 0x00,
  0x1f, 0xc1, 0xbc, 0x40, 0x00, 0x00, 0x00, 0x1f, 0xc0, 0x00, 0x42, 0x00, 0x00, 0x00, 0x1f, 0xc0,
  0x00, 0x40, 0x00, 0x00, 0x00, 0x10, 0x40, 0x00, 0x40, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3f,
  0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0x80,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x19, 0x80, 0x00, 0x00,
  0x00, 0x70, 0x00, 0x19, 0x80, 0x06, 0x00, 0x00, 0x50, 0x00, 0x09, 0x00, 0x00, 0x00, 0x00, 0x70,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x70, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x70, 0x00, 0x00,
  0x00, 0x22, 0x00, 0x00, 0x60, 0x00, 0x00, 0x00, 0x41, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x41,
  0x00, 0x00, 0x70, 0x00, 0x00, 0x00, 0x49, 0x00, 0x00, 0x70, 0x00, 0x00, 0x00, 0x41, 0x00, 0x07,
  0x77, 0x00, 0x00, 0x00, 0x22, 0x00, 0x1f, 0x27, 0xe0, 0x30, 0x00, 0x1c, 0x00, 0x36, 0x23, 0xa0,
  0x20, 0x00, 0x00, 0x00, 0x38, 0x20, 0xe0, 0x3e, 0x00, 0x00, 0x00, 0x10, 0x20, 0x40, 0x87, 0x00,
  0x00, 0x20, 0x00, 0x20, 0x00, 0x81, 0x00, 0x00, 0x00, 0x00, 0x20, 0x07, 0xe7, 0x00, 0x00, 0x00,
  0x00, 0x20, 0x0d, 0x6c, 0x00, 0x06, 0x40, 0x00, 0x00, 0x0f, 0x38, 0x00, 0x07, 0xc0, 0x00, 0x00,
  0x2f, 0xf0, 0x00, 0x07, 0xc0, 0x00, 0x00, 0x25, 0xe0, 0x00, 0x00, 0x07, 0xc0, 0x00, 0x14, 0xc0,
  0x00, 0x00, 0x0f, 0xe0, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x08, 0x20, 0x00, 0x86, 0x00, 0x00, 0x00,
  0x0b, 0x00, 0x00, 0x64, 0x00, 0x00, 0x00, 0x08, 0x20, 0x00, 0x34, 0x00, 0x00, 0x00, 0x0f, 0xe0,
  0x00, 0x18, 0x00, 0x00, 0x00, 0x07, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
const unsigned char IITDLogo [] PROGMEM = {
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0d, 0xd8, 0x00, 0x00, 0x00, 0x00, 0x00,
  0xdf, 0xff, 0x80, 0x00, 0x00, 0x00, 0x01, 0xff, 0xff, 0xc0, 0x00, 0x00, 0x00, 0x0f, 0xe0, 0x03,
  0xf8, 0x00, 0x00, 0x00, 0x0f, 0x80, 0x00, 0x78, 0x00, 0x00, 0x00, 0x3e, 0x00, 0x00, 0x1f, 0x00,
[... 430 lines cut ...]
    {
      n = n_sensor;
      NTP();
      display.drawBitmap(3, 20, co2Logo, 40, 40, WHITE);
      display.setTextSize(2);
      display.setCursor(47, 30); display.println(c);
      display.setTextSize(1);
      display.setCursor(53, 50); display.println(n);
      display.display();
      delay(2000);
      display.clearDisplay();
    }
    else
    {
      NTP();
      display.drawBitmap(3, 20, co2Logo, 40, 40, WHITE);
      display.setTextSize(2);
      display.setCursor(47, 30); display.println(c);
      display.display();
      delay(2000);
      display.clearDisplay();
    }
  }
}`
Can you help me with this?

paulvha, 2020-10-03:
It would have been handy to include some of the error messages to get a direction on your issue. I think you get messages that some variables/routines are double defined. If that is the case I suspect the reason is that in the file Adafruit_SSD1306.h, around line 41 it includes Wire.h. 

Change in Adafruit_SSD1306.h the line from: #include <Wire.h> to something like : #include <scd30/SoftWire/SoftWire.h>

regards,
Paul

PrateekGoyal18, 2020-10-03:
These are the errors:
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::available()':

C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\scd30-master\src\SoftWire/SoftWire.cpp:172: multiple definition of `TwoWire::available()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\Wire\src/Wire.cpp:239: first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::read()':

SoftWire.cpp:(.text._ZN7TwoWire4readEv+0x0): multiple definition of `TwoWire::read()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire4readEv+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::peek()':

SoftWire.cpp:(.text._ZN7TwoWire4peekEv+0x0): multiple definition of `TwoWire::peek()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire4peekEv+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::flush()':

SoftWire.cpp:(.text._ZN7TwoWire5flushEv+0x0): multiple definition of `TwoWire::flush()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire5flushEv+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::write(unsigned char)':

SoftWire.cpp:(.text._ZN7TwoWire5writeEh+0x0): multiple definition of `TwoWire::write(unsigned char)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire5writeEh+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::write(unsigned char const*, unsigned int)':

SoftWire.cpp:(.text._ZN7TwoWire5writeEPKhj+0x0): multiple definition of `TwoWire::write(unsigned char const*, unsigned int)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire5writeEPKhj+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::setClock(unsigned int)':

SoftWire.cpp:(.text._ZN7TwoWire8setClockEj+0x0): multiple definition of `TwoWire::setClock(unsigned int)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire8setClockEj+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::requestFrom(unsigned char, unsigned char, unsigned char)':

SoftWire.cpp:(.text._ZN7TwoWire11requestFromEhhh+0x0): multiple definition of `TwoWire::requestFrom(unsigned char, unsigned char, unsigned char)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire11requestFromEhhh+0x0): first defined here

Multiple libraries were found for "WiFiClientSecure.h"
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::requestFrom(unsigned char, unsigned char)':

 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFiClientSecure
Multiple libraries were found for "SPI.h"
 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SPI
SoftWire.cpp:(.text._ZN7TwoWire11requestFromEhh+0x0): multiple definition of `TwoWire::requestFrom(unsigned char, unsigned char)'

Multiple libraries were found for "NTPClient.h"
 Used: C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\NTPClient-master
Multiple libraries were found for "Adafruit_GFX.h"
 Used: C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\Adafruit_GFX_Library
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire11requestFromEhh+0x0): first defined here

 Not used: C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\Adafruit-GFX-Library-master
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFi
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::requestFrom(int, int)':

 Not used: C:\Program
SoftWire.cpp:(.text._ZN7TwoWire11requestFromEii+0x0): multiple definition of `TwoWire::requestFrom(int, int)'
[... 26 lines cut ...]
Multiple libraries were found for "FirebaseESP32.h"
 Used: C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\Firebase-ESP32-master
SoftWire.cpp:(.text._ZN7TwoWire17beginTransmissionEh+0x0): multiple definition of `TwoWire::beginTransmission(unsigned char)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire17beginTransmissionEh+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::beginTransmission(int)':

[... cut ...]
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:(.bss.Wire+0x0): first defined here

c:/users/prateek/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: Warning: size of symbol `Wire' changed from 304 in C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o to 16 in C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o

collect2.exe: error: ld returned 1 exit status

paulvha, 2020-10-03:
Thanks as I thought, it finds double definitions in both Wire and SoftWire. Make the change as I proposed. (I can't check as I do not have all your hardware and libraries installed)
regards
Paul

PrateekGoyal18, 2020-10-03:
I put the SoftWire folder from your library in Adafruit_SSD1306 library folder and added the line `#include <SoftWire/SoftWire.h>`
Still the same error:
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::available()':

C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\scd30-master\src\SoftWire/SoftWire.cpp:172: multiple definition of `TwoWire::available()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\Wire\src/Wire.cpp:239: first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::read()':

SoftWire.cpp:(.text._ZN7TwoWire4readEv+0x0): multiple definition of `TwoWire::read()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire4readEv+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::peek()':

SoftWire.cpp:(.text._ZN7TwoWire4peekEv+0x0): multiple definition of `TwoWire::peek()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire4peekEv+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::flush()':

SoftWire.cpp:(.text._ZN7TwoWire5flushEv+0x0): multiple definition of `TwoWire::flush()'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire5flushEv+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::write(unsigned char)':

SoftWire.cpp:(.text._ZN7TwoWire5writeEh+0x0): multiple definition of `TwoWire::write(unsigned char)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire5writeEh+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::write(unsigned char const*, unsigned int)':

SoftWire.cpp:(.text._ZN7TwoWire5writeEPKhj+0x0): multiple definition of `TwoWire::write(unsigned char const*, unsigned int)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire5writeEPKhj+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::setClock(unsigned int)':

SoftWire.cpp:(.text._ZN7TwoWire8setClockEj+0x0): multiple definition of `TwoWire::setClock(unsigned int)'

Multiple libraries were found for "NTPClient.h"
 Used: C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\NTPClient-master
Multiple libraries were found for "SPI.h"
 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SPI
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire8setClockEj+0x0): first defined here

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFi
 Not used: C:\Program
Multiple libraries were found for "FirebaseESP32.h"
 Used: C:\Users\Prateek\OneDrive\Documents\Arduino\libraries\Firebase-ESP32-master
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::requestFrom(unsigned char, unsigned char, unsigned char)':

Multiple libraries were found for "FS.h"
 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\FS
Multiple libraries were found for "SPIFFS.h"
SoftWire.cpp:(.text._ZN7TwoWire11requestFromEhhh+0x0): multiple definition of `TwoWire::requestFrom(unsigned char, unsigned char, unsigned char)'

 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SPIFFS
Multiple libraries were found for "HTTPClient.h"
 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\HTTPClient
Multiple libraries were found for "WiFiClientSecure.h"
 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFiClientSecure
Multiple libraries were found for "SD.h"
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire11requestFromEhhh+0x0): first defined here

 Used: C:\Users\Prateek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SD
 Not used: C:\Program
[... 27 lines cut ...]
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::beginTransmission(unsigned char)':

SoftWire.cpp:(.text._ZN7TwoWire17beginTransmissionEh+0x0): multiple definition of `TwoWire::beginTransmission(unsigned char)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:Wire.cpp:(.text._ZN7TwoWire17beginTransmissionEh+0x0): first defined here

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o: In function `TwoWire::beginTransmission(int)':

SoftWire.cpp:(.text._ZN7TwoWire17beginTransmissionEi+0x0): multiple definition of `TwoWire::beginTransmission(int)'

[... cut ...]
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o:(.bss.Wire+0x0): first defined here

c:/users/prateek/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: Warning: size of symbol `Wire' changed from 304 in C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\Wire\Wire.cpp.o to 16 in C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\libraries\scd30-master\SoftWire\SoftWire.cpp.o

collect2.exe: error: ld returned 1 exit status

PrateekGoyal18, 2020-10-03:
By any chance do you have any code snippet for integrating scd sensor with some other i2c device like bme sensor, oled etc?
That will work for me, I'll make the required changes.

paulvha, 2020-10-03:
The issue is still the same.. somewhere Wire.h is still included.. I see that in Adafruit_GrayOLED.h.  You need to make the same change (around line 33). THis is part of the Arduino-GFX library.

PrateekGoyal18, 2020-10-03:
There is no Adafruit_GrayOLED.h file in Arduino-GFX library folder. I also checked other files in the Arduino-GFX library, nowhere else is Wire.h included.

[image]

paulvha, 2020-10-03:
Weird.. I have just installed Arduino_GFX and the file shows up in mine.

Somewhere however Wire.h is still included. That is the root cause of the problem. Check all the code and includes.

An alternative to quickly check:  Goto the ESP32 boards Wire library( I am on Ubuntu so it would be something like  ~/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire/src). Rename Wire.h in this folder to Wire_org.h. Now try to compile (which should fail) and notice the path from which Wire.h is called.
Make sure to rename back after you have finished.

regards,
Paul

[image]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F15648366%2F94988699-cd213d00-058c-11eb-9bd5-96450df50a7e.png&data=02%7C01%7C%7C1d6095c51ebe4908d5c708d86782c2e2%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637373158589214267&sdata=1WXIpiTALFgKdvh3camHbOEgo1GsDz3cVzyD3I8TyYo%3D&reserved=0>

PrateekGoyal18, 2020-10-03:
@paulvha I commented out the wire.h library from my code, and it worked. Is it a fine solution?

paulvha, 2020-10-03:
didn't you get an error message ? I would expect the compile to fail as Wire.h could not be found.. can you share the compile verbose output ?

paulvha, 2020-10-03:
I have no re-read your post. As you commented out wire.h somewhere in your code that is the solution.!

closed 2020-10-03 by paulvha