paulvha/scd30 #7: No values from the SDC30

DirkScheumann, 2020-10-05:
Hi,
I used an Arduino and and SDC30 to make a CO2 measuring device. I used the Sparfun library and got it to work, but I couldnt really change the measuring interval. The sensor stops always working. 
With your library this seemed to work without any problems. I used your examples to change all kind of things.
The only thing which didn'twork anymore? I didn't get values for the CO2, temperature or humidity. The ony thing I got was "0" or "NAN"
I am new to this and don't really know how the libraries work, I opend yous and replaced some lines from the SparkFun_SCD30_Arduino_Library.ccp (line 189 to 278) in your ***.ccp. After I have done this I got values for the temparature and the humidity but not for CO2.
Any idea why I am not getting anything with you library?

-Dirk Scheumann

paulvha, 2020-10-06:
hi Dirk

Without knowing what you have changed it is hard to say. All I can say is that if I use the original library it works.
Maybe that is the best way forward. copy /paste the current library to another place and re-install the SCD30 library again.
If it then still fails enable debug :  #define scd_debug 2 and share the errors and debug information

regards,
Paul

DirkScheumann, 2020-10-06:
Hi Paul,
this is what I am getting if I use your library and example 1 with scd_debug 2:

19:18:47.119 -> SCD30 Example 1
19:18:47.119 -> NO ESP8266/ ESP32 expected
19:18:47.119 -> Begin measuring with pressure offset 0
19:18:47.119 -> sending to I2C address 0x61, Command 0x10 : COMMAND_CONTINUOUS_MEASUREMENT, arguments 0x0, CRC 0x81
19:18:47.119 -> set measurement interval 
19:18:47.119 -> sending to I2C address 0x61, Command 0x4600 : COMMAND_SET_MEASUREMENT_INTERVAL, arguments 0x2, CRC 0xe3
19:18:47.119 -> sending to I2C address 0x61, Command 0x5306 : COMMAND_AUTOMATIC_SELF_CALIBRATION, arguments 0x1, CRC 0xb0
19:18:47.119 -> sending to I2C address 0x61, Command 0xD033 : CMD_READ_SERIALNBR
19:18:47.175 -> 
19:18:47.175 -> Receiving: x 0, 0x32 x 1, 0x31 x 2, 0x1E x 3, 0x35 x 4, 0x30 x 5, 0x81 x 6, 0x35 x 7, 0x31 x 8, 0xB0 
19:18:47.175 -> SCD30 serial number : 215051R
19:18:47.175 -> get Firmware level
19:18:47.175 -> sending to I2C address 0x61, Command 0xD100 : CMD_GET_FW_LEVEL
19:18:47.175 -> 
19:18:47.175 -> Receiving: x 0, 0x03 x 1, 0x42 x 2, 0xF3 
19:18:47.175 -> SCD30 Firmware level: Major: 3	, Minor: 66
19:18:47.175 -> sending to I2C address 0x61, Command 0x202 : COMMAND_GET_DATA_READY
19:18:47.175 -> 
19:18:47.175 -> Receiving: x 0, 0x00 x 1, 0x01 x 2, 0xB0 
19:18:47.175 -> co2(ppm):sending to I2C address 0x61, Command 0x202 : COMMAND_GET_DATA_READY
19:18:47.175 -> 
19:18:47.175 -> Receiving: x 0, 0x00 x 1, 0x01 x 2, 0xB0 
19:18:47.175 -> sending to I2C address 0x61, Command 0x300 : COMMAND_READ_MEASUREMENT
19:18:47.227 -> 
19:18:47.227 -> Receiving: x 0, 0x44 x 1, 0xBF x 2, 0x1A x 3, 0x85 x 4, 0xE3 x 5, 0x47 x 6, 0x41 x 7, 0xD4 x 8, 0x3C x 9, 0x6A x 10, 0x11 x 11, 0x48 x 12, 0x42 x 13, 0x41 x 14, 0xDD x 15, 0x42 x 16, 0xC7 x 17, 0x01 
19:18:47.227 ->  CO2 : 0x85E3, Temperature 0xFFFF, Humidity 0x6A11
19:18:47.227 -> 0 temp(C):0.0 humidity(%):0.0
19:18:49.216 -> sending to I2C address 0x61, Command 0x202 : COMMAND_GET_DATA_READY
19:18:49.216 -> 
19:18:49.216 -> Receiving: x 0, 0x00 x 1, 0x01 x 2, 0xB0 
19:18:49.216 -> co2(ppm):sending to I2C address 0x61, Command 0x202 : COMMAND_GET_DATA_READY
19:18:49.259 -> 
19:18:49.259 -> Receiving: x 0, 0x00 x 1, 0x01 x 2, 0xB0 
19:18:49.259 -> sending to I2C address 0x61, Command 0x300 : COMMAND_READ_MEASUREMENT
19:18:49.259 -> 
19:18:49.259 -> Receiving: x 0, 0x44 x 1, 0xC0 x 2, 0xCC x 3, 0x7A x 4, 0xFE x 5, 0xC9 x 6, 0x41 x 7, 0xD4 x 8, 0x3C x 9, 0xA6 x 10, 0x48 x 11, 0xA0 x 12, 0x42 x 13, 0x40 x 14, 0xEC x 15, 0xD6 x 16, 0x21 x 17, 0xA8 
19:18:49.259 ->  CO2 : 0x7AFE, Temperature 0x0000, Humidity 0xA648
19:18:49.259 -> 0 temp(C):nan humidity(%):nan
19:18:51.261 -> sending to I2C address 0x61, Command 0x202 : COMMAND_GET_DATA_READY

It looks like I am getting something but without debug mode I get either "0" or "NAN". Any idea whats wrong?

-Dirk

paulvha, 2020-10-06:
hi

I see indeed the SCD30 is providing data. I'll try to connect my sensor this evening and let you know.
Is the done with the standard example1 as-it-is ?

Greetings from The Netherlands
Regards
Paul

DirkScheumann, 2020-10-06:
hi, it's done with the example as it is and the library as it is. I installed it new, like you suggested. I downloaded the library from here.

Dirk

paulvha, 2020-10-06:
HI Dirk,

I think I found the root cause of the issue and I am making the final test on the solution. The original code works without a problem on ESPXXXX, but it fails on Arduino.  I  will share tomorrow. It now works on Arduino, but need to test for ESPxxxx and other platforms ( SAMD, Apollo3 etc)

regards,
Paul

DirkScheumann, 2020-10-06:
Hi Paul, that's good news. I am really excited to test it tomorrow.

Good night!

Dirk

paulvha, 2020-10-07:
HI Dirk, I have posted an updated version which I have tested on Uno, ESP32 and Apollo3. The root cause was around translating the received bytes into a float. The old structure worked fine on ESP32, but failed on Uno and Apollo3. Let me know if we can close the issue
regards,
Paul

DirkScheumann, 2020-10-07:
Hi Paul, thanks a lot. It works! Soooo coool!

Did you ever try to change the measuring intervall while the sensor was working? Did you also encounter that the sensor stops working sometimes?

-Dirk

paulvha, 2020-10-07:
Thanks Dirk

No I have not changed the interval while it is working and I do get once about in every 10 times that data is not available.. but not that it does not it stop working.

Do you power on 5V ?

regards
Paul

DirkScheumann, 2020-10-07:
Hi, I power it with 5V. It doesn't work with 3,3V. It draws a very high current, maybe something around 300mA every time it measures.
I think it just stopped because I tried to run it on a 3,7V Lipo. It never stoppedn on 5V USB.

-Dirk

paulvha, 2020-10-08:
yes. that can be the issue. According to the datasheet it would take 75mA in measurement, but i have not measured that myself. As the original problem is solved, I will close the issue. Good luck !

closed 2020-10-08 by paulvha