paulvha/scd30 #8: TemperatureOffset

DirkScheumann, 2020-10-15:
Hi Paul,
I never see any changes if I put the airSensor.setTemperatureOffset((float) 6); line in my programm for the arduino. Is it possible there is a similar problem with the format of the numbers like there was with the readings I never got in my last post? The numbers change like crazy if I put numbers like 200 in the program but these numbers are not right  either. Do I have to put 10 in the program to change te reading by one degree?

-Dirk Scheumann

paulvha, 2020-10-15:
HI Dirk

Happen to be behind my computer.. and checked the source code:

 * August 2018: all this does for now is lower the SCD30 temperature reading to with the offset value * over a period of 10 min, while increasing the humidity readings. NO impact on the CO2 readings.

If it does not impact Temperature and/or Humidity, try to set with a uint16 instead of float

setTemperatureOffset(uint16_t tempOffset)

But now each tick is 0.01C so for 6 degrees adjustment sent 6 x 100 = 600.

regards,
Paul

DirkScheumann, 2020-10-15:
Hi, the float changed the temperature, but the humidity went tru the roof. Value always around 90.
The uint16_t doesn't change anything compared to no temperature offset. The sensor shows 29C and I would like to get it to 23C.
I have to write it like this: airSensor.setTemperatureOffset((uint16_t) 600);
or am I doing something wrong?

-Dirk

paulvha, 2020-10-16:
I have just checked. No issues here.  I have added an example4 to test. you can manually add the wanted degree change to the measured value ( e.g. 6 <enter> for 6 degrees). It will loop to display the SCD30 results and thus you can see the impact.

DirkScheumann, 2020-10-17:
Hi Paul, thanks for the program. I pt it on the Arduino and checked the reading of the sensor. It startet with 26C. I put in 3 and the behavior of the sensor changed the temperature went up. After 5 min the temperature was 29C. The problem is I didn't want to have 29C, I wanted to have 23C. I put in -3 ant the it changed the number to something wired and the temperature went down to -xxx degrees, very fast. Do you have another idea? Maybe I substract just 3 from the value I get from the sensor and show this number on the display on my Co2-, Temp-, Humidity-meter.

-Dirk

paulvha, 2020-10-17:
HI Dirk

You can NOT provide a negative value. The value is UNSIGNED as the SCD30 handle SIGNED. The compensation already negative to compensate for the measurement impact of components on the board that produce heat.

Have the SCD30 run for a while (say an hour) without compensation (set to zero). If the reading output is more or less stable then check what compensation is you want to apply.

As with many sensors you get a better result with having an airmovement along the sensor. A small fan as it does not need to be windforce 11. (I often use an old cooling fan of a processor)

regards,
Paul

DirkScheumann, 2020-10-18:
Hi, I tried several differnt things, but no success. It shows still 6 degrees more than all the other thermometers in the house on the balcony, wherever. As soon as I enter a number in the program example 4, the value goes very fast up, it never goes down. If enter 0 it still goes up and stays at whatever it was going to oustside. Very strange but it doesn't affect the CO2 reading. I put it outside it measures the CO2 somwhere around 450, 480, all good.
Any other ideas?

-Dirk

paulvha, 2020-10-19:
Hi Dirk

Co2 reading is NOT impacted by this change. That does not mean CO2 is not impacted by temperature, because it is, but that is NOT impacted by setting the offset. Only Temperature and humidity output results are only impacted.

The offset values are stored in NVRAM and thus available after power-up. After power-up first set to 0 and let it go up and stabilize for some time.

I have connected mine to the UNO, started example 4. The temperature around the sensor is 21.8 (external temperature meter), the SCD30 tells 22.98.
I have entered 0 and after about 370 seconds the SCD30 shows more or less stable on 25.4. The external temperature meter is showing 22.1.
So I enter 3 on 370 seconds and around 970 seconds (that is about 10 min) it shows 22.43. The external temperature meter is still showing 21.8.

regards,
Paul

DirkScheumann, 2020-10-25:
Hi Paul,

I tried it yesterday again an it worked without any problems. I did it the same way like you described in you post. I have set it to zero, and just waited until the reading was really stable. Then I set it to 6, because it was 6 degrees off. Until 10-15 minutes it was stable and now it work exactly the way I want.

Thanks for your help. Stay healthy. Covid numbers in the Netherlands are rising.

-Dirk Scheumann

paulvha, 2020-10-25:
Thanks Dirk for the update.

I have uploaded yesterday a new version of example4 . You can now adjust with a number behind the decimal poiint  eg. 6.2 instead of 6. I have also included a procedure in the top of the sketch to perform the calibration. Turned out I needed 3.1 offset.

I have also added an example5, where the data is displayed on an LCD 16 x2 (Sparkfun). You can set CO2 limit to trigger the display to turn red if above, or only show when it is above, even combine with button to only show normal for 10 seconds after pushing.

As for Covid.. we try to avoid it as much as possible, but I expect we will all going to get it. Many, if not most, without serious health issue, some with. But  we are very careful ..

Regards,
Paul
Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

DirkScheumann, 2020-10-25:
Hi, I used the sensor to build a CO2 meter. I used a 1,8" TFT screen that shows CO2, temperature and the humidity. It also changes the colors of the CO2 value from green to yellow to red. My wife works as a teacher and it helps her to determine if they have to open the windows or not. It gives her an idea how much used air is in the room. I hope it helps her not to get Corona, soon. I also think we get it all, but I hope I can get it if some kind of medication is developped.

-Dirk

paulvha, 2020-10-26:
Interesting to see we had the same thought.

Regards,Paul

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

paulvha, 2020-11-21:
issue has been solved

closed 2020-11-21 by paulvha