paulvha/scd30 #9: ESP32: Using Two I2C devices causing an compilation error

Akiiem, 2020-11-21:
I'm facing a compilation error always when I'm compiling paulbha_SDC30.h and all of the BME280 I2C library I tested so far. As far I understand the issue than because of the multiple use of wire.h. lib. Is there a chance getting a lib based on HW wire? working with multiple I2C devices? I understood there is a clock_streching issue with ESP32.
Thx.

paulvha, 2020-11-21:
Hi

The standard I2C on ESP32 is indeed not using clock stretching which is the root cause of all this. The Softwire.h can be used for all devices that want to use I2C. Hence in the top of example10, around line 79 I give example of what to change in case of using the SParkfun  BME280 library, but something like this can be applied to others. I have seen that next to a change in the header file ( .h), it sometimes also needs a change to the code file (.cpp) where they #include <wire.h> 

I am not owning the standard ESP32 library but even then..if I remember well it was an ESP32 hardware limitation. The best way to overcome is take the bit-bang version of the ESP8266.

paulvha, 2023-01-26:
closing old issues

closed 2023-01-26 by paulvha